KaufHA/PLF12 #3: Wrong Power reporting

robertsorensen, 2023-02-15:
One of my Kauf PLF12 plugs is reporting power usage despite the plug is turned off and despite current is reported to be 'zero'.
Please see the attached plot from HA.
The daily energy from reported from the plug is 'zero'.
What could be the reason for this?

[image]

bkaufx, 2023-02-16:
That's a weird one.  It's just the one plug?  Has to be some kind of hardware issue but hard to imagine what would cause that.  Maybe the power monitoring chip on it is bad.  If you email me your address I can send a replacement.

robertsorensen, 2023-02-17:
No worries Brian. I just received 4 more so I'll see if the same pattern occurs.
FWIW - sometimes during the night it stopped reporting erroneous power numbers.

and then - with the new plugs - I get this error message during compilation:
[image]
and I can't access the device with the ip-address - like the web-server doesn't work.

robertsorensen, 2023-02-18:
So FWIW - I believe the above error message was caused  by the fact that I was running the latest HA but not the latest ESPHome (I was on the last from 2022). I tired to compile other ESPHome project that works/worked and got the same error message. I then updated to ESPHome 2023.2.1 and now it compiles. But -
I got code onto one unit and it behaves weird as per the log. Also - the relay doesn't toggle at all and stays in off position. Not even when I plug in the plug with Boot State set to always on.
I also had significant problems updating OTA from within ESPHome - downloading timed out - sometimes after 10%, sometimes 43%, sometimes more and eventually I updated from the webserver with a .gz file.
Also - not that reported power increase despite voltage and current is '0' and the plug is off

Got 3 more plugs to play with :-) 

[image]

robertsorensen, 2023-02-18:
Another update from the 2nd of the 4 devices.
I verified that the device was working and I could toggle the relay from the button and from the UI (Webserver).
I then compiled a new binary to get my own device name and my static IP address in and uploaded the .gz file using the webserver. I now see the same issues with this device - can't toggle or turn on and I do get similar JSON related error message and some bogus power reporting ......
I wonder if something significant has changed in ESPHome 2023.2.x or whether you changed anything in you YAML file.

My YAML (Sorry - I couldn't get it to format)

`substitutions:
  name: raindance-id-plug-06
  device_description: Raindance Indoor Plug 06 KA
  comment: Raindance Indoor Plug 06 KA
  friendly_name: Indoor Outlet 6

packages:
  Kauf.PLF12: github://KaufHA/PLF12/kauf-plf12.yaml
  wifi: !include common/wifi_static_ip.yaml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  comment: ${device_description}

#Configure IP address
wifi:
  manual_ip:
    static_ip: !secret raindance_id_plug_06_ip`

BostonEnginerd, 2023-02-18:
There's some kind of issue with ESPHome 2023.2.  All of the PLF12 switches I have started malfunctioning when updated to that version of ESP home. They all showed crazy power readings and the relays stopped working. Reverting to the stock firmware fixed the issue.

bkaufx, 2023-02-18:
OK.  I think the order of the package imports needs to be reversed with the new ESPHome update.  I'll check it out now.

bkaufx, 2023-02-18:
Alright, I just pushed an update to reverse the order of the packages.  Should be working now.  You might (probably will) have to manually remove the .esphome/packages folder to get the new version.

robertsorensen, 2023-02-18:
Thanks for the quick turn-around time Brian.
I just updated one of my plugs and it is now working as expected.

closed 2023-02-18 by robertsorensen