KaufHA/PLF12 #6: PLF12 runs out of memory and reboots frequently

neillbell, 2023-02-21:
ESPHome version: 2023.2.3

I've seen this now with two PLF12s, so I don't think that it is a hardware problem.

I just did a fresh install of a PLF12. The install goes well, but after a couple of toggles of the relay, the device appears to run out of memory and will frequently disconnect and/or reboot itself.

I've attached the log of the install. The log shows the disconnects and memory allocation failures ("Could not allocate memory for JSON document"). From looking at my DHCP log, I can see that it also rebooted near the end of this log.

I also have a PLF10. I have updated it to ESPHome 2023.2.3 and have not seen the same problem.

[logs_kauf-plug-7aaab1_run.txt](https://github.com/KaufHA/PLF12/files/10795548/logs_kauf-plug-7aaab1_run.txt)

bkaufx, 2023-02-21:
As far as I've been able to tell this is an issue with the web_server.  I haven't been able to figure out what exactly the issue is other than possibly having more entities than an esp8266 can handle.  Is this with kauf-plug-minimal.yaml?  If that still has problems, try copying the yaml file into your local file and removing the `web_server:` line.

neillbell, 2023-02-21:
This is with kauf-plf12.yaml. I've seen the same behavior with kauf-plf12-lite.yaml.

kauf-plf12-minimal.yaml works great.

neillbell, 2023-02-21:
At your suggestion, I tried copying both kauf-plf12-lite.yaml and kauf-plf12.yaml locally and removing the web_server: functionality.

They both work great! The only unusual thing I see is an occasional line like this:

`[11:41:32][W][api.connection:071]: 10.0.110.27: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11`

bkaufx, 2023-04-11:
> I have a similar problem on my PLF12 plugs - at least some of my 12 plugs.
> On one of the plugs where I saw the problems I wasn't able to change
> firmware with using ESPHome OTA method. This was consistent with what I saw
> when I initially setup all my plugs. I had to use the OTA method in the
> WebServer when I initially configured the plugs. This time however I wasn't
> even able to use the WebServer OTA method! Both method resulted in time-out
> during the update.
> On a second plug I was able to update with WebServer OTA and could then
> install the minimal version (copied locally) however after that I noticed
> that voltage reporting was off - despite all the substitutions values being
> the same!
>
> Next step - hopefully tomorrow is to revert back to the "full" YAML but
> stored locally and see if this brings back correct voltages and he ability
> to do normal OTA update but I'm obviously a bit reluctant to remove the
> 'webserver:' entries as it so far has been the only way I managed to update
> the plugs.
>
> Is anything being investigated on the KaufHA side? At least based on my
> observation the problem should be relatively easy to reproduce.

ESPHome added API encryption by default, which increases the bin file side
quite a bit. I played around with the yaml to see if I could reduce the
size significantly without losing functionality but couldn't get there. For
now you need to either remove API encryption or temporarily flash with
kauf-plug-minimal.yaml and then go back to kauf-plug.yaml with API
encryption. I have more detailed instructions on the common repo readme
page under troubleshooting. Let me know if that doesn't do it for you.

robertsorensen, 2023-04-12:
First couple of tries on a couple of my plugs are positive.
Even the voltage levels reverted back to the correct values

robertsorensen, 2023-04-12:
I properly spoke too fast.
What I did observe is that by downloaded (using Webserver) a compressed binary of the kauf-plug-minimal.yaml I was subsequent able to upload the full yaml from the ESPHome dashboard.
I have NOT been using encryption at all nor have I ever used password with my api: entry so I haven't really changed anything in my full yaml compared to what it was before and what apparently prevented me from uploading from the dashboard. But again - I now appear to able to update my device OTA.
I still see the 'Could not allocate memory for JSON document'; perhaps a bit less but never the less I see it from time to time. 
Also - and this is new with the recent upload.
When I toggle the plug in HA I hear the relay toggle (turn off) and I see the small slider in HA moving from on to off. However after ~2 seconds the slider in HA moves back to the 'on' position though the relay remain in the off position.
If I press the slider again - I see the indicator moving to the off position and again after ~2 seconds it moves back to the on position without the relay toggling at all.
If however I press the slider and press it again before it moves from the off to the on position in HA then the relay toggles back on.

Just for completeness this is the YAML that I'm using for my plugs - they should look familiar.
(I swear it's pasted with CMD-e)

`substitutions:
  name: "raindance-id-plug-01"
  device_description: Raindance Indoor Plug 01 KA
  comment: Raindance Indoor Plug 01 KA
  friendly_name: Indoor Outlet 1
  device_static_ip: !secret raindance_id_plug_01_ip

  

packages:
  Kauf.PLF12: github://KaufHA/PLF12/kauf-plf12.yaml
  wifi: !include common/wifi_static_ip.yaml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  comment: ${device_description}
  project:
    name: "Kauf.PLF12"
    version: "2.02(m)"`