KaufHA/PLF12 #7: PLF12 not enough space for OTA

juichi, 2023-04-09:
With newly delivered PLF12s, in ESPHome (2023.3.1), I changed the name, friendly name, and use_address.  Compilation worked fine, but then I got a notice that the binary file was too large for OTA.  

To get around this I changed the package to kauf-plf12-lite.yaml, flashed the plug, then I changed the repository back to kauf-plf12.yaml.  This worked.

Attempting attempting to reflash I get this error with the kauf-plf12.yaml.  

`ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)`

Here are the compilations size messages:

`INFO Uploading /data/counteraquariumlight/.pioenvs/counteraquariumlight/firmware.bin (630064 bytes)`
`INFO Compressed to 431897 bytes`

Additionally visiting the local ip of the lights does not generate a webserver, I get a file download dialog from chrome.   Weird.

juanuribeo13, 2023-05-23:
Thanks @juichi for sharing the workaround. I was getting the same error but in my case had to go with the `kauf-plf12-minimal.yaml` package.

bkaufx, 2023-05-23:
Yea you are right it should be mentioned in the plug readme.  I'll add it.  Removing the API encryption should fix this.  I'm looking for ways to slim down the binary enough to work with API encryption without losing functionality.

steveomg, 2023-06-07:
If you want to keep encryption you can use minimal.yaml and gzip it then upload through the interface.  You lose all the Kauf options, but you still get current, power, and basic plug functions.

bkaufx, 2023-06-07:
I reconfigured how the LEDs are controlled to take up less space.  So the main kauf-plf12.yaml should be working now with API fine.

steveomg, 2023-06-07:
Confirmed - OTA works fine now.

closed 2026-10-29 by KaufHA