KaufHA/PLF12 #8: No webserver running after joining wifi network?

Famicoman, 2023-05-01:
Hi, I purchased this device with the sole intention of triggering it via terminal with curl, and do not want to integrate it with any home automation hub/software.

A review on Amazon mention being about to send POST requests to a plug with crafted URLs, and this seems to be backed up with info on the common repo here, https://github.com/KaufHA/common

However, when I try this on my end (substituting the proper IP address) it seems like curl can't access the webserver on the plug. When I nmap'd it, there wasn't anything running on port 80. Am I using the wrong port? If not, does the default firmware not allow this? 

Thanks,
Mike.

bkaufx, 2023-05-01:
Maybe try rebooting the plug. Sometimes the ESPHome webserver won't be up after the captive portal is used until the plug is rebooted.  It should be working as you describe.  If you put the IP address in a web browser it should bring up a web interface.

Famicoman, 2023-05-01:
Solved, thanks so much!

closed 2023-05-01 by Famicoman