KaufHA/PLF12 #9: PFL12 Red/Blue Flash - offline now

PDXUser, 2023-05-29:
This is my ESPhome YAML file for this plug:
```

---
substitutions:
  name: dishwasher
  friendly_name: Dishwasher
  disable_entities: "false"
packages:
  Kauf.PLF12: github://KaufHA/PLF12/kauf-plf12.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
wifi:
  ssid: wifi_ssid
  password: wifi_password
  
# Optional manual IP
  manual_ip:
    static_ip: 192.168.1.209
    gateway: 192.168.1.254
    subnet: 255.255.255.0
```

I had changed the names and wanted to disable entities then add the manual IP.  It seemed to flash okay, but now not online anymore.  I checked the light and it is alternating between red/blue.  Is there a hard reset which I can do to factory reset by holding the power button down?  I tried this and it only turned off red flashing or turning on and doing the same thing red/blue - I can't find it in the phone's SSID list now.

bkaufx, 2023-05-29:
So you got the one plug back that you were having trouble with?  

Well, the issue I see with your yaml is that the wifi credentials aren't substitutions.  So the plug is probably trying to connect to a wifi network called "wifi_ssid" that I'm guessing doesn't exist.

Needs to be like:

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

PDXUser, 2023-05-30:
I got the primary plug I was having issues back by doing the reset by holding down the power button for 30 seconds.  I guess it disassociated itself with the WiFi and kept the rename to "dishwasher" so when I found it in the SSID lists to connect my phone to, it was under the "dishwasher" hostname which was fine.  

I think we can close this issue though it's up to you. :)  I really love how your plugs work so far!   Easy as pie to install in Home Assistant!