KaufHA/PLF12 #11: Can you add a wifi signal sensor to be showen on HA

jbrandek, 2023-06-20:
I tried to add this to the config file but then the device become unavailable

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

Thanks

bkaufx, 2023-06-20:
It's working fine for me to add it in my local yaml config.  What is happening exactly when it becomes unavailable?

jbrandek, 2023-06-20:
had really bad connection and lose connectivity had to keep trying to flash back without it
but i will try again

jbrandek, 2023-06-20:
INFO ESPHome 2023.5.5
INFO Reading configuration /config/esphome/plug-tv-balcon.yaml...
INFO Detected timezone 'America/Panama'
INFO Generating C++ source...
INFO Compiling app...
Processing plug-tv-balcon (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/plug-tv-balcon/.pioenvs/plug-tv-balcon/src/esphome/components/wifi_signal/wifi_signal_sensor.cpp.o
Compiling /data/plug-tv-balcon/.pioenvs/plug-tv-balcon/src/main.cpp.o
Linking /data/plug-tv-balcon/.pioenvs/plug-tv-balcon/firmware.elf
RAM:   [=====     ]  47.6% (used 39020 bytes from 81920 bytes)
Flash: [======    ]  58.8% (used 602317 bytes from 1023984 bytes)
Building /data/plug-tv-balcon/.pioenvs/plug-tv-balcon/firmware.bin
esp8266_copy_factory_bin(["/data/plug-tv-balcon/.pioenvs/plug-tv-balcon/firmware.bin"], ["/data/plug-tv-balcon/.pioenvs/plug-tv-balcon/firmware.elf"])
========================= [SUCCESS] Took 10.50 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.107.213
INFO Uploading /data/plug-tv-balcon/.pioenvs/plug-tv-balcon/firmware.bin (606464 bytes)
INFO Compressed to 417679 bytes
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.107.213 using esphome API
WARNING Can't connect to ESPHome API for 192.168.107.213: Timeout while connecting to ('192.168.107.213', 6053)
INFO Trying to connect to 192.168.107.213 in the background

jbrandek, 2023-06-20:
substitutions:
  name: plug-tv-balcon
  friendly_name: Plug Tv Balcon

  disable_webserver: "true"
  disable_entities: "false"

packages:
  kauf.plf12: github://KaufHA/PLF12/kauf-plf12.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.107.213
    gateway: 192.168.107.1
    subnet: 255.255.255.0

# Enable Home Assistant API
api:
  encryption:
    key: "vWBgSbM+9VawMy+9pXR+Uy4HNuWFCjdCuxIypsBOhBg="

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s

jbrandek, 2023-06-20:
Sorry, its working now it took a while but its working

Thanks

closed 2023-06-20 by jbrandek