KaufHA/PLF12 #14: Error Loading into ESPHome

jablack10716, 2023-07-20:
INFO Reading configuration /config/esphome/kauf-plug-8218ac.yaml...
Failed config

external_components: [source /config/esphome/.esphome/packages/a0579e33/kauf-plug.yaml:149]
  - [source /config/esphome/.esphome/packages/a0579e33/kauf-plug.yaml:149]
    source: 
      type: git
      url: https://github.com/KaufHA/common
      ref: v2023.07.18
    
    Unknown value 'y', valid options are 'us', 'microseconds', 'ms', 'milliseconds', 's', 'sec', 'seconds', 'min', 'minutes', 'h', 'hours', 'd', 'days'.
    refresh: never [source /config/esphome/.esphome/packages/a0579e33/kauf-plug.yaml:153]

bkaufx, 2023-07-20:
What version of ESPHome are you using?  This was an ESPHome bug that was fixed a long time ago.  Our configs require ESPHome to be up to date.

jablack10716, 2023-07-20:
OK.  I will uninstall whatever version of ESPHome I have and re-install.  See if that fixes the issue.

jablack10716, 2023-07-20:
That was the issue.  I guess I was using a very old version of ESPHome that it wouldn't even update.  Thanks for the quick response.

closed 2023-07-20 by jablack10716