KaufHA/PLF12 #18: Connection reset by peer / plug seem to reboot every minute.

streamnsight, 2023-08-31:
I keep getting dozens of warning messages in the logs from all 4 plugs I just got.

```
2023-08-30 15:44:09.598 WARNING (MainThread) [aioesphomeapi.connection] kauf-plug-7aafad @ 10.1.10.108: Connection error occurred: [Errno 104] Connection reset by peer
```

In the logbook, I see the plugs seem to reset every minute for some reason, yet the metrics and everything 'looks' normal.

These are stock firmware, no edits, just plugged in and configured WiFi.
What's going on and how to troubleshoot this?
Thanks

bkaufx, 2023-08-31:
Is the plug rebooting or just losing connection to Wi-Fi?  What does the uptime sensor show? (this can be viewed from the web interface or enabled in Home Assistant)

Try updating for starters.  You can import into the ESPHome dashboard or download the bin.gz file and update through the web interface.  Here is the file you need: https://github.com/KaufHA/PLF12/releases/download/v2.052/kauf-plf12-v2.052-update.bin.gz

Here's where someone had similar issues and there are a few ideas here: https://github.com/esphome/issues/issues/1237

Possibly something weird with compatibility between your router and ESPHome.  Do you have any other ESPHome devices that work fine?

Maybe try another router to connect the plugs to or try flashing with Tasmota just to see if the problems go away.  Don't have anything plugged into them when you flash Tasmota because they are going to turn on and off a lot at first until you program in the right template.  https://github.com/KaufHA/PLF12#recommended-tasmota-template

streamnsight, 2023-08-31:
Thanks for the prompt response. The uptime sensor says it's been up for several hours, so I guess it's not a 'reboot' but maybe WiFi related. My WiFi router has the same SSID for 2.4GHz and 5GHz, so maybe that's the issue? I need to investigate further.

streamnsight, 2023-08-31:
I tried the fixes mentioned in the linked issue (switching to bandwidth 20MHz and setting a fixed channel) but that did not help.
What did help was change the DHCP lease time from 1min to 5min and after that it doesn't seem to disconnect anymore.

closed 2023-08-31 by streamnsight

bkaufx, 2023-08-31:
Interesting.  Thanks for the info.