KaufHA/PLF12 #19: ERROR Error binary size: Error: ESP does not have enough space to store OTA file.

mharleydev, 2023-09-06:
I'm new to ESP Home and Kauf. I looked at closed issues but didn't see anything.

I'm trying to update from 2023.8.2 to 2023.8.3. This is the first update I'm trying since buying the plugs. I'm trying to update via the ESP Home add-on in Home Assistant. Below is the end snippet and will also attach the full log file.

How can I resolve this issue? Is there a suggested update strategy for ESP Home in general as it relates to Kauf devices?

```Linking /data/kauf-plug-7aab42/.pioenvs/kauf-plug-7aab42/firmware.elf
RAM:   [=====     ]  48.0% (used 39348 bytes from 81920 bytes)
Flash: [======    ]  59.0% (used 604209 bytes from 1023984 bytes)
Building /data/kauf-plug-7aab42/.pioenvs/kauf-plug-7aab42/firmware.bin
esp8266_copy_factory_bin(["/data/kauf-plug-7aab42/.pioenvs/kauf-plug-7aab42/firmware.bin"], ["/data/kauf-plug-7aab42/.pioenvs/kauf-plug-7aab42/firmware.elf"])
========================= [SUCCESS] Took 34.69 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of kauf-plug-7aab42.local
INFO  -> 192.168.1.74
INFO Uploading /data/kauf-plug-7aab42/.pioenvs/kauf-plug-7aab42/firmware.bin (608368 bytes)
INFO Compressed to 419185 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```
[logs_kauf-plug-7aab42_run.txt](https://github.com/KaufHA/PLF12/files/12536583/logs_kauf-plug-7aab42_run.txt)

StevenBrehmer, 2023-11-13:
I have ran into this with each of my configurations. 

my work around has been to flash them with the min.yaml then reflash with the full... ( tedious, so id love to see a better solution)

bkaufx, 2023-11-13:
The strategy in general is to remove api encryption from your local config then you never have to worry about this again.  I had this fixed for the plugs, they must have made a change to ESPHome in 2023.8.3 that increased the size again.  I will take another look.

mharleydev, 2023-11-27:
Thanks.

For anyone who may find this by searching, removing the encryption, as Brian said, seems to have fixed my update issues. See [Binary Size Error](https://github.com/KaufHA/common/blob/main/README.md#binary-size-error) for details on disabling encryption.

wittelw, 2023-12-26:
Thanks to all. I found configuring the yaml packages with the following gives me all the functionality I need without having to switch to min.yaml and back or remove api encryption. Results may vary (this for Firmware: 2023.12.5), and may not work as codebases increase in size:
`Kauf.PLF12: github://KaufHA/PLF12/kauf-plf12-lite.yaml`

imagesafari, 2023-12-30:
After commenting out the api encryption key, Home Assistant doesn't seem to want to communicate with the device, throwing an error:

Reading failed: BAD_INDICATOR errno=11

How do I get around this?

closed 2026-10-29 by KaufHA