KaufHA/PLF12 #20: [Question] Hypothetically, what happens if pushed over max amperage?

PDXUser, 2023-09-14:
I know the PLF12's are rated for 15A Load, 1800W General use.    I have a Roper Washer (RAX6144EQ1) whose plate says 9.8A on the machine for power draw, and it lives on it's own dedicated breaker in the box.   Assuming I don't have any extra strain on the  motor which would draw a heaver amperage, it should be okay.  But what happens if the machine were to exceed the ratings for some oddball reason?  

Will the plug melt, catch fire, sing the national anthem?  Or would it shut down safely and refuse to work anymore?

bkaufx, 2023-09-14:
98% it would just burn out the relay and you wouldn't have any idea it happened except that the plug will be stuck either on or off.  I think the chance of anything catastrophic happening is pretty low as long as you stick to normal 120v stuff around the house.

PDXUser, 2023-09-14:
Thanks!  I figured you would be the best person to ask since you made it!  Closing the issue. :)

closed 2023-09-14 by PDXUser