KaufHA/PLF12 #21: device not listed in esphome

flyingtech, 2023-11-11:
I had to restore my home assistant server.  The restore point was before getting the new plugs (4 days ago).  now I'm having issues with kauf plugs. 

the two devices did not show up in ESPhome to be adopted. but they are showing up as new devices to be integrated.  I integrated the first with no problems, but the other is asking me for an encryption key.
[image]

But neither device is showing up in the esphome ad-on.

I'm lost??????

bkaufx, 2023-11-13:
What you would do to manually add into the dashboard is create a yaml file and paste the following code in. Then:

- change IP address to IP address of device.
- change name and friendly_name as desired.
- after you flash once, remove use_address line

```
substitutions:
  name: kauf-plug
  friendly_name: Kauf Plug

packages:
  kauf.plf12: github://KaufHA/PLF12/kauf-plf12.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  use_address: 192.168.1.12

```

flyingtech, 2023-11-13:
Thank you for the help, this fixed the issue.  Really love these plugs and supporting a small business is icing on the cake!

closed 2023-11-13 by flyingtech