KaufHA/PLF12 #22: Error binary size: not enough space to store OTA file using ESPHome 23.11.1

Imlbrown, 2023-11-16:
When updating the PFL12 plug based on the ESPHome 2023.11.1 firmware released today. I receive the following error:

Error binary size: Error: ESP does not have enough space to store OTA file.

Output size of firmware.bin is 610880 bytes and compressed to 420785 bytes.

Hope that is enough info.

Thanks,
Larry

Imlbrown, 2023-11-17:
I did see the response from a previous similar issue and it said a work around was to disable api encryption, but since the size issue, using api encryption, was corrected between that version of ESPHome and this current version, I didn't know if you were going to attempt a fix that would work with api encryption enabled.

bkaufx, 2023-11-17:
Yea I will try to figure it out but not sure how quick I will be able to get to it.  I would recommend just removing api encryption, but if you want to keep it you can hobble along by going through the -minimal.yaml package temporarily every time you update or you could also just move to minimal permanently.  The main yaml doesn't add that much that you couldn't implement yourself with much more basic yaml.

Imlbrown, 2023-11-17:
I can just disable the encryption for now, just wanted you to be aware.

No worries and no rush.

Thanks,
Larry

wildfire305, 2023-11-27:
I managed to get the firmware for 2023.11.1+ to work with api enabled by outputting the bin file and compressing it manually with 7zip using the gzip option. This made it smaller than esphome output, and thus it was able to fit within the size needed.

closed 2026-10-29 by KaufHA