KaufHA/PLF12 #25: issue connecting to wifi.

Jagsta2121, 2024-01-21:
I just received my first plug from amazon i followed the instructions to connect to wifi. it just sits there flashing red once i connect it. i have tried making a 2.4ghz network only disabled band steering even setup a different ap. it cannot get it to connect. it takes the settings because it stops broadcasting its initial network. i have tried updating it with the latest bin file. i'm not finding it in the wifi clients or on the dhcp server. any tips would greatly be appreciated.

bdkuhman, 2024-01-22:
I had a similar issue; Initial network setup worked fine, adopted into esphome and flashed the updated yaml, but then they wouldn't connect, reproduced with a second one.

turned out that `wifi_ssid` and `wifi_password` were incorrect in my `secrets.yaml` from playing around a while back, so had to spin up a new wifi network for them to connect to, and then I was able to flash an updated file; though I did have to disable encryption due to #22

bkaufx, 2024-01-22:
Hold the button for 30 seconds and it should re-initialize to the default credentials (initial_ap / asdfasdfasdfasdf).  You can stop holding the button when the blue light starts blinking.  You can try re-entering the credentials and maybe you just entered them wrong last time.  You can also try flashing a custom firmware with your credentials built in.  If the flashing process isn't working through the plug's AP, you can create a temporary hotspot with the default credentials and the plug will connect to it.  Flashing should be more reliable that way.