KaufHA/PLF12 #26: Power calculation

MrMinecraft97, 2024-02-01:
Question - How is the power value calculated? Is it just a simple volts * amps or is there power factor stuff involved?

If I draw 0.5 amps from the plug at about 122V, it only shows a power of ~35 watts, even though it should be ~60 watts. Power is going into a laptop charger. Should I just change the "Kauf Plug Scale Power" value to adjust this?

Thanks

Also, on an unrelated note, the version number on the web interface shows "2.06(y)" while it was OTA flashed with 2.052. Not sure if this is a bug or not.

bkaufx, 2024-02-02:
The power monitoring chip on the plug gives 3 different values for current, voltage, and power.  They can all be separately calibrated so one of them is probably off.  Thanks for letting me know about the version number.  It might be wrong in the binary file.  I'll check it out.

MrMinecraft97, 2024-02-10:
Got it. Thanks.

closed 2024-02-10 by MrMinecraft97