KaufHA/PLF12 #27: Compilation Error with ESPHome 2023.10.0

siretart, 2024-02-11:
Trying the recommended configuration:

```yaml
substitutions:
  name: bedroom-lamp
  friendly_name: Bedroom Lamp

packages:
  kauf.plf12: github://KaufHA/PLF12/kauf-plf12.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
results in::

```
INFO ESPHome 2023.10.0
INFO Reading configuration /config/esphome/kaufplug1.yaml...
ERROR Unable to load component esp8266:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 165, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/data/external_components/e00f7cee/components/esp8266/__init__.py", line 32, in <module>
    from .gpio import PinInitialState, add_pin_initial_states_array
  File "/data/external_components/e00f7cee/components/esp8266/gpio.py", line 146, in <module>
    pins.gpio_base_schema(
AttributeError: module 'esphome.pins' has no attribute 'gpio_base_schema'
Failed config

esp8266: [source /data/packages/a0579e33/kauf-plug.yaml:100]
  
  Component not found: esp8266.
  board: esp01_1m
  restore_from_flash: True
  early_pin_init: False
  start_free: 76
  global_addr: global_forced_addr
  ```

bkaufx, 2024-02-12:
Update esphome to latest.

siretart, 2024-02-12:
thanks, that indeed did the trick.

closed 2024-02-12 by siretart