KaufHA/PLF12 #28: Updates fail after ESPHome2024.3.0 is updated.

Node815, 2024-03-20:
ESPHome introduced a breaking change https://esphome.io/changelog/2024.3.0#ipv6  to now look for ipv6 addresses which has broken the update on my Kauf12 plugs.  I do not have IPV6 on my network.   After updating to version 2024.3.0 when I run the plug updates, I get the following error:

```src/esphome/components/network/util.cpp: In function 'esphome::network::IPAddresses esphome::network::get_ip_addresses()':
src/esphome/components/network/util.cpp:47:41: error: 'class esphome::wifi::WiFiComponent' has no member named 'get_ip_addresses'; did you mean 'get_ip_address'?
   47 |     return wifi::global_wifi_component->get_ip_addresses();
      |                                         ^~~~~~~~~~~~~~~~
      |                                         get_ip_address
*** [.pioenvs/washingmachine/src/esphome/components/network/util.cpp.o] Error 1
========================== [FAILED] Took 2.65 seconds ==========================```

Shocktrooper, 2024-03-20:
I just got this as well not sure if it was deprecated previously and changed now or it was just changed in the latest update

bkaufx, 2024-03-20:
Sorry, the ESPHome release came out late last night my local time so I didn't get around to updating our components until this morning.  Everything should be working for the plugs now.  You'll either have to delete the .esphome/packages directory or wait 24 hours for your ESPHome instance to decide it wants to check for package updates again.

Node815, 2024-03-20:
I tried your solution of deleting the directory and it still failed.

![image](https://github.com/KaufHA/PLF12/assets/20736534/776ede81-d679-4019-8f87-78bff2a73395

[logs_update-all.txt](https://github.com/KaufHA/PLF12/files/14670181/logs_update-all.txt)

Shocktrooper, 2024-03-20:
Where is the .esphome directory located? I was unable to find it on the server

Node815, 2024-03-20:
I used the VScode server (Studio Code Server) plugin in Home Assistant to get to the directory and then expand it to show the .esphome which is a hidden directory. From within should be a packages directory.   Also, you can find this in the Home Assistant Config directory if you are using Home Assistant.  I never specified I was using that in my original post here so this may or may not apply to your question.

bkaufx, 2024-03-20:
Is the .esphome directory still empty after you tried updating?  If so then maybe they are using a different directory now for the home assistant plugin.  In that case you just deleted old data that was left over from an old version and not the active packages directory.

Looks like that is the case:
https://www.reddit.com/r/Esphome/comments/18e69jp/where_are_the_generated_files_stored/
https://community.home-assistant.io/t/access-esphome-file-system-to-retrieve-elf-build-file/548973

bkaufx, 2024-03-20:
Those links explain how to get to the directory or you can just wait 24 hours.

Node815, 2024-03-20:
Neither in either directory from the link, also I logged into the ESPHome container to see if it was there and could not find it. 

I'm seriously considering reflashing to Tasmota, would that offer the same energy monitoring and stats that the ESPHome integration does?  I do like the early publish as well for monitoring my NAS and Desktop, but can live without that but rely on the rest for automations.

And if I should decide on a later date, flash back to the ESPHome Firmware?

Shocktrooper, 2024-03-20:
@Node815 The update came out yesterday night...  Considering that the patch came out within 24 hours you are better off with this firmware than tasmota

Shocktrooper, 2024-03-21:
@bkaufx Do we have to do what the error says to do here or is there a different way we should be updating?

```========================= [SUCCESS] Took 10.05 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.1.22
INFO Uploading /data/build/kauf-plug-c62123/.pioenvs/kauf-plug-c62123/firmware.bin (616560 bytes)
INFO Compressed to 423158 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)```

Node815, 2024-03-21:
All 4 of mine updated with out issues this morning. I don't know if this is related to your above issue @Shocktrooper

bkaufx, 2024-03-21:
I'd just remove the API encryption and re-add into home assistant after updating.  You do this by commenting out the API section, or at least the encryption part of it.  You could also add -minimal to the yaml file (change from kauf-plf12.yaml to kauf-plf12-minimal.yaml), update, then remove -minimal and update again, if you wanted to keep API encryption.

Shocktrooper, 2024-03-27:
That worked for me! I think this issue can be closed btw

closed 2024-04-11 by Node815