KaufHA/PLF12 #29: Syntax for POST request via REST API on PLF12 plug?

tonylane26, 2024-04-20:
"Firmware version 2.056(f) made using ESPHome version 2023.7.1"  (BTW the latest on github is v2.052)

The documentation on the ESPhome REST API indicates that a SWITCH should have a "/turn_on" method e.g.
/switch/dehumidifier/turn_on  (also presumably /turn_off)

Copying the format from the plug via the /events page I came up with:
http://kauf-plug-xxxxxx.local/switch/kauf_plug/turn_on

From my browser this returns: {"id":"switch-kauf_plug","value":false,"state":"OFF"}
and the plug does not turn on.

Is there a list of GET and POST commands available for this Plug? I had also wanted to use:
http://kauf-plug-xxxxxx.local/sensor/kauf_plug_device_in_use
which also shows in the Events Source API but doesn't work on the REST API.

I don't want to use Home Assistant as I already have everything on node-red. If this doesn't work I might try re-flashing the plug and using Tasmota. I'd prefer to use them with the web server, straight out of the box.

Thanks in advance.

TJNII, 2024-04-24:
For my PLF12:

Turn on: `curl [device]/switch/kauf_plug/turn_on -X POST`
Turn off: `curl [device]/switch/kauf_plug/turn_off -X POST`

I haven't figured out anything else yet, I was struggling to figure out the `device` and `id` components before I found this post.   (Can they be added to the UI?  Is `curl [device]/switch` expected to return a 500 or 404 depending on trailing `/`?)

bkaufx, 2024-04-25:
If you update to the latest ESPHome version by compiling yourself, I believe they no longer require http post.  So you can just use normal URLs in a web browser and commands should work.  I'll try to release a new update binary soon with it.

The in_use sensor is a binary_sensor so you need to change sensor to binary_sensor in your URL. Let me know if that doesn't work.

The documentation for the ESPHome web api all applies to our products so the documentation there may help: https://esphome.io/web-api/index.html