KaufHA/PLF12 #30: Occasional Spurious Power readings using REST API

tonylane26, 2024-04-23:
Since I started logging power from a sump pump with the PLF12 last week, I've been getting spurious Power readings, not more than once an hour. Sometimes it will go 6-8 hrs without a glitch, but mostly 10-20 times a day.
I am using the REST API (out of the box), reading every 5s via node-red. I tried filtering for valid messages (status code 200) so I'm sure these are coming from the PLF12 itself.
A normal pump reading should be about 750 W, but these glitches always return a value of exactly 85.60712 W, too constant to be a real load, or even leakage.
For pump running status I am now filtering out anything below 100W, but this glitch also affects the Daily kWh reading coming from the plug. 
I am thinking of switching to Tasmota (see issue #29), but it looks like that can only send MQTT every 10s (fastest setting), which is too slow. 
Any suggestions?

bkaufx, 2024-04-25:
Can you update with the ESPHome dashboard?  If so, I'd recommend using the lite or minimal yaml file instead of the default so you aren't using our custom power monitoring component.  I have seen this issue with sustained high power draw but I thought I had it fixed.  I'll take another look but for now I'd recommend switching to the stock ESPHome version.  If you can't compile your own I can try to post a precompiled binary that will work.

tonylane26, 2024-04-27:
I installed ESPhome on a Linux partition on my Chromebook. I could not find a way to install the ESPhome dashboard without installing Home Assistant. Also I couldn't figure out how to make ESPhome connect to the Kauf Plug. It seemed to want at least one device connected via serial (USB). 

I thought about trying to install ESPhome on a Wemos D1 mini, just to get it all up and running, but this was taking me away from where I wanted to go.

tonylane26, 2024-04-27:
This is certainly not a high power application. The pump is about 750W and only runs occasionally. The reason it's so noticeable is the fact that it's normally zero, and then kicks in with the same 85.60712 W reading every time.
 
Can the ESPhome stock edition you mentioned, be flashed using your out of the box firmware? If so can you point me to a suitable version?

If I went to the ESPhome stock version, what functionality would I lose compared to the OoTB version?

Thanks