KaufHA/PLF12 #35: standard config appears broken as of esphome 2024.5.5 update

mbeynon, 2024-06-10:
After updating esphome to 2024.5.5, all Kauf plug updates fail with this compilation error.  Looks like something changed upstream in Kauf code or the Kauf provided .yaml.  Any ideas?

Compiling .pioenvs/kauf-plug-3/lib67b/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pioenvs/kauf-plug-3/lib67b/ESP8266WiFi/ESP8266WiFi.cpp.o
Archiving .pioenvs/kauf-plug-3/libaf0/libHash.a
Compiling .pioenvs/kauf-plug-3/lib67b/ESP8266WiFi/ESP8266WiFiAP.cpp.o
/data/packages/a0579e33/kauf-plug.yaml: In lambda function:
/data/packages/a0579e33/kauf-plug.yaml:332:7: error: 'wifi_wificomponent' was not declared in this scope; did you mean 'wifi_wificomponent_id'?
  332 |           wifi_wificomponent->set_ap_timeout(2147483647);
      |       ^   ~~~~~~~~~~~~~~
      |       wifi_wificomponent_id
Compiling .pioenvs/kauf-plug-3/lib67b/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Archiving .pioenvs/kauf-plug-3/libace/libESPAsyncTCP-esphome@1.2.3.a
Compiling .pioenvs/kauf-plug-3/lib67b/ESP8266WiFi/ESP8266WiFiGratuitous.cpp.o
Compiling .pioenvs/kauf-plug-3/lib67b/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling .pioenvs/kauf-plug-3/lib67b/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
Compiling .pioenvs/kauf-plug-3/lib67b/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
*** [.pioenvs/kauf-plug-3/src/main.cpp.o] Error 1
========================= [FAILED] Took 91.53 seconds =========================

bkaufx, 2024-06-10:
This was changed in our config when 2024.5.0 came out.  You must be stuck on an old version of the yaml. Try cleaning build files or manually deleting the .esphome/packages folder.  If nothing else, reinstall the esphome add-on.

mbeynon, 2024-06-12:
Cleaning build files worked.  Thank you.

closed 2026-10-29 by KaufHA