KaufHA/PLF12 #36: Plug won't turn power off

synthc, 2024-06-17:
Communication with the plug seems to be working fine and I can turn it on and off both wirelessly and manually via the button (and the LED indicates the appropriate state); but the power is stuck on - it will never turn off the device that's plugged into it.

I've tried doing a full reset and re-adding it to the network and it worked for a little while before starting this behavior again.  Now it won't work no matter what I do.

bkaufx, 2024-06-18:
Can you hear the relay clicking?  Pretty weird for it to work and suddenly not work.  Sounds like a bad relay.  E-mail me and I can send a replacement.  brian@kaufha.com

synthc, 2024-06-18:
Yes, I hear it clicking.  I sent you an email, but your email server rejected it:

>[<brian@kaufha.com>](mailto:brian@kaufha.com): host mx1.improvmx.com[13.37.195.136] refused to talk to me:
    551-5.7.1 Your IP is black listed by SpamCop.org (See
    https://www.spamcop.net/bl.shtml). 551-5.7.1 To best protect our users from
    spam, 551 5.7.1 we will ignore all incoming emails from you until the
    matter is resolved.

bkaufx, 2024-06-20:
That's a pretty weird error.  Email is working for me.  Maybe try sending from a different email address or computer.

synthc, 2024-06-20:
I tried again, the email didn't bounce this time.

synthc, 2024-06-25:
The replacement is working.  Thanks.

synthc, 2024-07-06:
This issue is has now started occurring with the replacement plug.  Exact same issue, I hear the relay click when I switch the plug on/off but it doesn't cut the power to what's plugged into it.

I have a single LED tube shop light plugged into it.  The plug itself is plugged directly into an outlet (I had it on a power strip and moved it to the outlet, thinking maybe the strip was interfering somehow).

I have three other PLF12's and all of them work just fine.  Do you have any idea what might be going on here?  If I turn the switch on/off repeatedly it eventually turns the light off.

bkaufx, 2024-07-10:
What are the specs of the light that is plugged into it?  Sounds like the symptoms you might expect if you were plugging in a large inductive load that created an inrush current and welded the relay open.

synthc, 2024-07-11:
120V; 0.72A; 42W; 60Hz