KaufHA/PLF12 #38: When updating Kauf plug PLF12 get error: 'wifi_wificomponent' was not declared in this scope

wittelw, 2024-07-02:
From the log file (attached) I get the following error trying to update my Kauf plug PLF12 from HA's ESPHome menu (ESPHome 2024.6.4):

```
Compiling .pioenvs/kauf-plug-821851/lib67b/ESP8266WiFi/ESP8266WiFi.cpp.o
/data/packages/a0579e33/kauf-plug-lite.yaml: In lambda function:
/data/packages/a0579e33/kauf-plug-lite.yaml:309:7: error: 'wifi_wificomponent' was not declared in this scope; did you mean 'wifi_wificomponent_id'?
  309 |           wifi_wificomponent->set_ap_timeout(2147483647);
      |       ^   ~~~~~~~~~~~~~~
      |       wifi_wificomponent_id
```
My kauf-plug-821851.yaml is:

```
substitutions:
  name: kauf-plug-821851
packages:
  Kauf.PLF12: github://KaufHA/PLF12/kauf-plf12-lite.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: <redacted>

ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
This config was compiling fine until a few ESPHome versions ago. Note I had to add the ota platform to my configs this last ESPHome version to avoid other errors that also cropped up in other ESPHome devices, but then got back to this same blocking error as before.

I've noticed from [searching KaufHA](https://github.com/search?q=owner%3AKaufHA%20wifi_wificomponent&type=code) that kauf-plf12-factory.yaml (as well as kauf-bulb-factory.yaml) have not been updated with the '_id' suffix in the 'wifi_wificomponent' lamdas.

I tried editing kauf-plf12-factory.yaml locally (HA Terminal into the Raspberry Pi) in /homeassistant/esphome/.esphome/packages/c5646bd4 and 'Clean Build Files' before 'Install', but this still produces the same error.

[logs_kauf-plug-821851_run (1).txt](https://github.com/user-attachments/files/16060902/logs_kauf-plug-821851_run.1.txt)

bkaufx, 2024-07-10:
You are stuck on an old version of the yaml I think.  This has been happening to a lot of people.  Change the package yaml file to PLF12/config/kauf-plf12-lite.yaml (add /config).  That may trick ESPHome into downloading the new version.  I moved it so that needs to be done anyway.  If ESPHome can't see the yaml file at that location, you'll probably need to reinstall the ESPHome add-on.

wittelw, 2024-07-10:
Thanks! Changing the packages line to "Kauf.PLF12: github://KaufHA/PLF12/config/kauf-plf12-lite.yaml" fixed the problem without having to reinstall anything, and the OTA upload was successful.

closed 2024-07-10 by wittelw