KaufHA/PLF12 #39: Possible to use with sump pump?

rocketraman, 2024-07-16:
Hello, I am interested in monitoring my sump pump on/off cycles with the PFL12.

Is it possible to use the plug with a sump pump? I have [this one](https://www.pentair.com/en-us/flow/hydromatic/hydromatic-products/hydromatic-residential-products/hydromatic-sump-pumps/hydromatic-w-d-v-a1-cast-iron-sump-pumps.html?queryID=2db4345237a2cf8ea9656c1c3b5dcab0&objectID=https://www.pentair.com/en-us/flow/hydromatic/hydromatic-products/hydromatic-residential-products/hydromatic-sump-pumps/hydromatic-w-d-v-a1-cast-iron-sump-pumps.html&indexName=prod_pentair_products).

I know monitoring large resistive loads can sometimes be difficult for relays.

A related question is: what is the failure mode of the plug? If the plug's smart features stop working for some reason, will the load continue to work?

bkaufx, 2024-07-29:
I wouldn't really recommend using it for a sump pump unless you can wire up a backup power source.  I'd hate for your house to flood if the plug failed.  If the smarts failed it might stay powered but if the relay failed it might not.  I would not trust it to do anything in particular on failure.

rocketraman, 2024-07-29:
Yeah that is the conclusion I came to as well. Thank you.

closed 2024-07-29 by rocketraman