KaufHA/PLF12 #40: Kauf Plug PLF12 Quit Working This Morning

PKCubed, 2024-09-15:
I had a Kauf Plug monitoring power you my server rack UPS. It had been measuring about 10 amps for days with no problems. This morning it shut off and my UPS went in to battery mode. After further investigation, plugging it in results in the following symptoms.

- The red light will flash shortly, sometimes it will stay on.
- The plug makes strange buzzing noises (frequency seems to changes when I touch it)
- No WIFI, no functionality at all, no power through the plug.

My best guess is that the power supply inside the plug supplying the ESP module, relay, and LEDs failed and isn't supplying sufficient power to the plug. From my experience, power supplies make bad noises when they fail.

Could this be a result of a high current load through the plug for a long period of time (10 amps in my case)? Or is this just a bad plug?

JollyRgrs, 2024-12-01:
I am also interested in this. I was just trying to look up the amp load max and could not find it on the website or here. I know the Sonoff S31 is good for 15A, but most smart plugs are only good for 10, so if your load is "about" 10 (I assume that might be an under/over number), it could potentially surpass the rated amperage. but if it just blew a fuse, I would expect it to just be dead. This is where my understanding on the hardware side ends and would love if there was a good understanding of this. If it is a physical limit, I'd like to understand so I don't push my plug past that limit.

bkaufx, 2024-12-03:
10A should be fine.  I don't think a constant load would stress the power supply too much.  That current really only goes through the relay so there's no reason the power supply would have to provide more current.  The only thing that would change is the detection circuitry has to switch faster, which admittedly would use more power from the power supply but it should be pretty minimal.

If you email me I will send a replacement.