KaufHA/PLF12 #42: Home Assistant ESPHome Compiler failing for PL12 plug

poodad, 2024-12-10:
Trying to use the Home Assistant ESPHome compiler to update some PL12 plugs. I get the following error

INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/kauf-plug-7aab31.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1014, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1091, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 945, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 935, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 772, in validate_config
    config = do_packages_pass(config)
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/packages/__init__.py", line 177, in do_packages_pass
    package_config = _process_base_package(package_config)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/packages/__init__.py", line 101, in _process_base_package
    repo_dir, revert = git.clone_or_update(
                       ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 90, in clone_or_update
    age = datetime.now() - datetime.fromtimestamp(file_timestamp.stat().st_mtime)
                                                  ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/pathlib.py", line 1014, in stat
    return os.stat(self, follow_symlinks=follow_symlinks)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/data/packages/c5646bd4/.git/HEAD'

ESPHome's yaml file for the plug is

substitutions:
  name: kauf-plug-7aab31
packages:
  Kauf.PLF12: github://KaufHA/PLF12/kauf-plf12.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: N9I/tlVlYM0oiSxPoyCPVjWP+rS287nOwROE0AkbLEA=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

I suspect something is wrong with this github file: github://KaufHA/PLF12/kauf-plf12.yaml

bkaufx, 2024-12-10:
I think the issue is on your end.  Maybe try cleaning build files.  You might have to back up all your yaml files and remove and reinstall the ESPHome add-on in Home Assistant.

poodad, 2024-12-10:
> I think the issue is on your end. Maybe try cleaning build files. You might have to back up all your yaml files and remove and reinstall the ESPHome add-on in Home Assistant.

I removed the ESPHome add-in and removed /config/esphome/*

Same results.

My PL10 plugs all work fine in ESP Home.

closed 2024-12-12 by poodad