KaufHA/PLF12 #43: Conflict with button integration in kauf-plf12-minimal.yaml

neillbell, 2024-12-10:
I was happy with the functionality provided by kauf-plf12-minimal.yaml, but I wanted to add a restart button. I modified my configuration to include:
```
button:
  - platform: restart
    name: $friendly_name Restart Firmware
    entity_category: diagnostic
```
This caused the following error:
```
binary_sensor.gpio: [source /data/packages/c5646bd4/kauf-plf12-minimal.yaml:98]
  platform: gpio
  
  ID 'button' conflicts with the name of an esphome integration, please use another ID name.
  id: button
```
I was able to get it working by making my own copy of kauf-plf12-minimal.yaml and changing the problematic name to something other than "button". It seems like the name in kauf-plf12-minimal.yaml should be changed since there is a "button" integration.

bkaufx, 2024-12-10:
Interesting thanks.  I'll probably change it to main_button or something.

bkaufx, 2024-12-10:
Ok I didn't realize this was only an issue on the minimal since I already had to fix this on the other ones.  I renamed it to button_in for you.

closed 2024-12-11 by neillbell