KaufHA/PLF12 #44: Failed update with newest ESPHome version

martan1912, 2025-02-20:
Hello,
Would you mind taking  look at this thread? There are issues with the latest ESPHome update, can't update Kauf PLug device.

https://community.home-assistant.io/t/esp-home-failed-update-on-a-device/850404

INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/kauf-plug-2.yaml…
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
File “/usr/local/bin/esphome”, line 8, in
sys.exit(main())
^^^^^^
File “/esphome/esphome/main.py”, line 1049, in main
return run_esphome(sys.argv)
^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/main.py”, line 1027, in run_esphome
config = read_config(dict(args.substitution) if args.substitution else {})
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/config.py”, line 1095, in read_config
res = load_config(command_line_substitutions)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/config.py”, line 949, in load_config
return _load_config(command_line_substitutions)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/config.py”, line 939, in _load_config
return validate_config(config, command_line_substitutions)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/config.py”, line 835, in validate_config
target_platform = core_config.preload_core_config(config, result)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/core/config.py”, line 222, in preload_core_config
if _is_target_platform(domain):
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/core/config.py”, line 184, in _is_target_platform
if get_component(name, True).is_target_platform:
^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/loader.py”, line 205, in get_component
return _lookup_module(domain, exception)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/esphome/esphome/loader.py”, line 181, in _lookup_module
module = importlib.import_module(f"esphome.components.{domain}")
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “/usr/lib/python3.11/importlib/init.py”, line 126, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File “”, line 1206, in _gcd_import
File “”, line 1178, in _find_and_load
File “”, line 1149, in _find_and_load_unlocked
File “”, line 690, in _load_unlocked
File “”, line 940, in exec_module
File “”, line 241, in _call_with_frames_removed
File “/data/external_components/e395f423/components/wifi/init.py”, line 106, in
cv.Required(CONF_STATIC_IP): cv.ipv4,
^^^^^^^
AttributeError: module ‘esphome.config_validation’ has no attribute ‘ipv4’

bkaufx, 2025-02-20:
I already updated the components and yaml configs to reference the update components. If you can delete the .esphome/packages folder, then your esphome instance will immediately check for updates and work fine. Otherwise, your ESPHome instance won't check again for 24 hours.

martan1912, 2025-02-20:
Thank you ! All good now.

closed 2025-02-20 by martan1912