KaufHA/PLF12 #45: Add id: tag to total_daily_energy

jcsouthworth, 2025-04-03:
Hello,
I use home assistant and I use a Kauf PLF12 to track power consumption by some hardware. I am trying to be more efficient with data storage and this plug is a base-ish load, so I get a datapoint every 0.001 kWh, which can lead to thousands of data points per day.

I think this is as easy as adding "accuracy_decimals:" or filter: delta to the total_daily_energy tab so I only update every 0.01 or 0.1 kWh as appropriate, but since there isn't an id: tag, I can't use the home assistant "extend:" functionality.sensor:
```
 - id: !extend total_daily_energy_sensor_id
   accuracy_decimals: 2
```

could we update the total_daily_energy field found in https://github.com/KaufHA/PLF10/blob/main/kauf-plug.yaml so that the total daily energy sensor has an id, like this:

  ```
- platform: total_daily_energy
    name: $friendly_name Total Daily Energy
    power_id: wattage
    filters:
        - multiply: 0.001  ## convert Wh to kWh
    unit_of_measurement: kWh
    state_class: total_increasing
    forced_hash: 1903527169
    forced_addr: 6
    global_addr: global_forced_addr
    id: new_total_daily_energy_sensor_id
```
if there is a good reason not to do this, let me know. I couldn't think of a good reason not to do this to allow easier end-user extension of this sensor/entity.

bkaufx, 2025-04-03:
done.  [`id: tde`](https://github.com/KaufHA/PLF10/blob/main/kauf-plug.yaml#L556)

jcsouthworth, 2025-04-03:
Your fix works perfectly. Thank you!

closed 2025-04-03 by jcsouthworth