KaufHA/PLF12 #46: Use Threshold missing

kflott, 2025-05-05:
I was previously using the Device In Use with some automation in HA.  After a recent update, I can no longer find the "Use Threshold" in the config and "Device In Use" seems to stay on all the time.  How can I add this back to my config?

kflott, 2025-05-05:
Reuploaded binary and updated and it came back after I recompiled in ESPHome.

closed 2025-05-05 by kflott

bkaufx, 2025-05-05:
Sorry I took this out of the default yaml package to conserve space.  ESPHome keeps getting bigger and bigger, and with the default ESPHome config now including API encryption I have to scrimp everywhere I can.  I meant to document the changes I made but life got crazy and I just haven't gotten around to it.  Sorry.

You can set the threshold in yaml with the substitution:

`  sub_threshold: "3"`

Alternatively, I'll add a new yaml that you can use to get all the config entities back in.  It just won't be the default.