KaufHA/PLF12 #47: Is it possible to flash back to espHome from Tasmota?

TheCraiggers, 2025-05-17:
I'm migrating away from MQTT and to more of a espHome setup as I've got a few more of the little buggers now. Looking online for info on migrating back does not give me much hope, although people do seem to occasionally do it with other devices. These are the two main things that are scary:

1. It's recommended to flash tasmota_minimal first so there's room on the 8266. However, it's explicitly mentioned in your manual to never flash this since the only method of updating is OTA. It looks like my version of tasmota supports gzipped uploads, so that could be an option if it wasn't for...

2. The espHome documentation on this mentions this: "WARNING! Migrating via OTA on ESP32, from Tasmota v12 (with boot partition) and up is currently not possible! Trying it could soft-brick your device!" As I'm currently on v13.4.0 of tasmota, it appears I'm hosed.

So, am I just screwed? I'm still extremely new to this whole esp32 thing, so I thought I'd ask here. Thank you in advance.

trashcatt, 2025-06-24:
Hey, so I was able to go back to espHome. 
What I did was update using TasmoAdmin (not sure if this actually matters).

I went to the upgrade page and uploaded the tasmota minimal firmware where it says MINIMAL Firmware and the kauf espHome firmware in the Full Firmware spot. Clicked next and let it flash the minimal firmware. (This is why I'm not sure if you need to use TasmoAdmin instead of the normal WEBUI)

After that I went to the webui of the switch and flashed the espHome firmware and everything worked as expected. 

You could probably just flash the minimal through the webui and then update to esphome using said webui.

Hope this helps.

trashcatt, 2025-06-24:
So after I flash the minimal firmware with TasmoAdmin, I then get this webui. 

[image]

Which allows me to flash the kauf esphome firmware.

TheCraiggers, 2025-10-17:
Just wanted to say thanks, @trashcatt. Worked perfectly, and your posts gave me the courage to ignore the documentation. :)